pbix/HA-solark-PV #3: Value names missing in config flow

poldim, 2021-09-24:
The form that pops up when adding the integration is missing the names of the fields.

[image]

As a comparison, here's the one for adguard: https://github.com/home-assistant/core/blob/master/homeassistant/components/adguard/config_flow.py

[image]

pbix, 2021-09-24:
Great feedback thanks.  Let me work on it this weekend and get back to you with an update.   I am a bit of a novice on writing custom integrations myself.   Meanwhile you are completing the form correctly so everything should work correctly.

pbix, 2021-09-25:
I tried to reproduce your results.  Here is what I see in my browser:

[image]

So I cannot reproduce this result.  I looked at your reference adguard and I do not see anything different that from what I did.  The displayed text in both cases is coming from the file "strings.json".  Did you use the same browser to create both of those screen shots?  I wonder if this could be one of the annoying rendering incompatibilities.  As far as I know (which is not much) there is nothing I can do inside a custom component to change anything about the way things are displayed.  This is all under the control of the HA environment.  I will Google around a bit to see if I can find anyone else with a similar issue but am otherwise not sure what I can do.

poldim, 2021-09-25:
Very odd.  I mainly use Chrome but also tried this with Safari on Mac and iOS within HA app which IIRC should be using same-ish webkit rendering engine.  Using the default theme during the daytime. 

[image]

[image]

[image]

pbix, 2021-09-26:
I reproduced your result with Chrome.  I found that if I flushed my browser cache the text appeared.  But as all the text was not visible in any browser due to its length I shorted these descriptions.  And I also made 'localhost' the default for the host address as this makes this descriptive font smaller and more fits.  Weird browser juju I don't enjoy.

I think it is improved, thanks for your input on this.  I am closing this issue.

[image]

[image]

closed 2021-09-26 by pbix