pbix/HA-solark-PV #6: Total PV Energy?

poldim, 2022-01-16:
I was curious to see what my total pv energy is now that my system has been online for a few months. I enabled this entity earlier today and got this:
[image]

Is this working correctly for you and I need to troubleshoot on my end?

poldim, 2022-01-16:
It looks like this [key](https://github.com/pbix/HA-solark-PV/blob/a24bc3a9a395fb4c424317ab098bc70a4fe24898/custom_components/solark/const.py#L111) goes to this [data](https://github.com/pbix/HA-solark-PV/blob/0ab2defb322a0c0f1cc2bb9a0d68cd7e3b98c003/custom_components/solark/hub.py#L134)

I'm not sure my comparison is correct:

[image]

But when I poll my own Solark, I get negative values for DayActive PowerWh and Total_Active_PowerWh low word (my system doesnt export)

```
modpoll 3.9 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2020 proconX Pty Ltd
Visit https://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: MODBUS/TCP, FC3
Slave configuration...: address = 1, start reference = 61, count = 20
Communication.........: 10.30.30.33, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave...
[61]: -10
[62]: 0
[63]: 0
[64]: -818
[65]: -1
[66]: 0
[67]: 0
[68]: 0
[69]: 0
[70]: 0
[71]: 8
[72]: 0
[73]: 6098
[74]: 0
[75]: 5844
[76]: 0
[77]: 54
[78]: 0
[79]: 13840
[80]: 6002
```

pbix, 2022-01-16:
Hmm.  I have been using that from the beginning without issue.  Here is a screen shot from in HA in which you can see this value displayed properly on my system.  Could you check the same on yours?  Have update to the the head BTW?

[image]

pbix, 2022-01-17:
I think I have determined that what is being displayed in that entity is actually grid port energy.  This I believe is the energy that has passed through the grid breaker on the inverter.  I found total PV energy at register 97.  Does it make sense to you that your grid port breaker has imported a small amount of energy?  Perhaps you have your loads connected to the load breaker and never export to your house?

In my case I found that register 97 data matches what is displayed on the front panel in the Solar Total field.  So what is in github now is not correct.

I would like to confirm your situation before I commit a fix this week.

poldim, 2022-01-18:
I'm not sure what either of these do or why my values are negative.  I initially ran my system with battery only for a few months.  And a few months ago, I added a small 5 panel array.  Inverter is configured to never export power to the grid and push all of the excess power in to the batteries.

[image]

I can't find anything in the sol ark monitoring app that matches the `-77`. `2.7` matches my `solark_daily_inverter_energy` which is in line with how you have it, but I can't find anything in the app showing 2.7 that would confirm it's inverter energy.  

You appear to be in the right direction but for me, 96 is the low word and 97 is the high word.  I did some more comparisons in the app, plain English descriptions on the right:
[image]

pbix, 2022-01-19:
I believe I have this correct at the head now.  Please update and report your results.

poldim, 2022-01-20:
Thanks, that fixed it.
[image]

closed 2022-01-20 by poldim