pbix/HA-solark-PV #7: solark_total_grid_breaker_energy has state class total_increasing, but its state is not strictly increasing

poldim, 2022-02-14:
I've got the below error a few times.  

I think you're reading registers 63 & 64 (totalgrid_e -> total_grid_breaker_energy). 
[image]

Here is my grid vs this measurement for the same time period.  It looks like what you're calling `total_grid_breaker_energy` decreases as I use grid power.  I don't export any power so this shouldn't be decreasing. So I'm not sure that the title makes sense? Or something else is wrong in my system...

[image]

[image]
 

```
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:311
Integration: Sensor (documentation, issues)
First occurred: February 13, 2022, 12:00:39 AM (1 occurrences)
Last logged: February 13, 2022, 12:00:39 AM

Entity sensor.solark_total_grid_breaker_energy from integration solark_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 10.0 (10.1) with last_updated set to 2022-02-13T07:55:15.428637+00:00. Please report it to the custom component author.
```

pbix, 2022-02-16:
The sensor total_grid_breaker_energy is the energy passing through the grid breaker of the SolArk inverter.  It will grow positive when power is exported to your home and negative when power is taken from your home (to charge the batteries for example).  Looks like you are in the net exporting power to your home from your SolArk but sometimes you are drawing power from your home.

The problem we have is confusion between "Grid Breaker" and "Utility".  The name "Grid" is printed on the breaker and shown in the manual.  That is why I selected this name for this sensor total_grid_breaker_energy.

Unfortunately I previously have a sensor total_grid_power which is the utility power so this is inconsistent and not helping things.

So these two sensors are not consistently named.  So I propose to rename the first sensor to "Total Utility Power".  Another possibility would be "Limiter Sensor Power" as the phrase "Limiter Sensors" appears in the manual.

Finally, I am using the state class "total_increasing" for these enegy sensors I should switch to "total" as it can go up and down depending on what you are doing with your SolArk.  Hopefully will get a chance to look at it this weekend.

poldim, 2022-03-02:
I missed the notification for your reply so I'm just finally replying now...

> The sensor total_grid_breaker_energy is the energy passing through the grid breaker of the SolArk inverter. It will grow positive when power is exported to your home and negative when power is taken [...]

My system is grid tied but is configured to disable any energy export (ever), it's actually configured to constantly draw a minimum 15w.  Since my SA is fed from a non critical loads panel, it could be possible that at night the SA is pushing power up through the grid breaker to reduce this consumption.  However, that should only be possible if my grid usage was ~ zero at that time.  And in the graph below, you can see there is constantly 300-400w of import from the utility. 

[image]

> The problem we have is confusion between "Grid Breaker" and "Utility". The name "Grid" is printed on the breaker and shown in the manual. That is why I selected this name for this sensor total_grid_breaker_energy.

I agree this is terrible naming.  It's probably an issue with translation from Chienese.  IMO, you should label it clear to their function rather than propegating bad naming. Maybe replace grid with input for something like *UTILITY CT POWER* and *INPUT POWER*, *INPUT CURRENT*.  
 
> Unfortunately I previously have a sensor total_grid_power which is the utility power so this is inconsistent and not helping things.

The graph below compares solark_total_grid_breaker_energy to a very good industrial meter that has CTs in the same location as the SA's external CTs.  They are reporting wildly different readings.

[image]

> So these two sensors are not consistently named. So I propose to rename the first sensor to "Total Utility Power". Another possibility would be "Limiter Sensor Power" as the phrase "Limiter Sensors" appears in the manual.

Another data point to consider is their app.  I agree that Grid here is still misleading but I do like how they separate out internal current transformers vs the external current transformer measurements.

[image]

> Finally, I am using the state class "total_increasing" for these enegy sensors I should switch to "total" as it can go up and down depending on what you are doing with your SolArk. Hopefully will get a chance to look at it this weekend.

Yup

pbix, 2022-03-05:
Not sure what to do here as there are a lot of sensors defined and the naming is simply not consistent today.  Not in my custom_ component or any of the SolArk documents I have seen,  

The sources in this inverter are:

Physical Source       Potential Name Prefix
PV                            PV
Batteries                  BATT
Grid Breaker            INV
Gen Breaker            GEN
Load Breaker           LOAD
External CTs            UTIL

Each of these sources has power, energy, currents, voltages and sometimes frequencies defined.  That is as many as 30 sensors I guess.  I have avoided the use of the term "GRID" above as it is ambiguous.  There term UTIL could be METER as this is shown in the manual often.

Any change in the names of these sensors at this point would cause backwards compatibility issues with any cards or automations that use them.

So not sure what to do on any renaming.

pbix, 2022-03-06:
For now I have changed several of the sensors declared as TOTAL_INCREASING to TOTAL.  This should resolve the main issue here, however, the ambiguity of these names remains.  If someone wants to propose a solution I am willing to listed.  For now take care as the term GRID is not used consistently in this component.

closed 2022-03-06 by pbix

poldim, 2022-03-08:
> Grid Breaker -------- INV

I agree with your other sensors names but this doesn't make sense as the inverter isnt a source itself.  I'd recommend INP for INPUT whatever that may be.

Not using grid in many places as you did was a good decision - just because they use the wrong names it doesn't mean you should propagate the problem.  Sol Ark isn't putting any effort into translating what the registers provide.  I will try pinging Synsunk since I think they write the actual firmware

I'm still scratching my head about what the register tied to solark_total_grid_breaker_energy is actually showing.  It shows energy decreasing and I can't think of a single sensor that should be doing that in my system since I'm never exporting.

I think a breaking change on the names would be ok as long as it's done one.  Fixing automations is no sweat, the bigger issue will be figuring out how to copy the data in influxdb so that you can leverage the last ~6 months of data.

pbix, 2022-03-08:
In my system the "grid" breaker on the inverter is the source of all the energy my system generates.  From that break my whole house is supplied.  So I do not understand your comment that it isn't a source and I don't get why you want to call it INPUT.  What is it an "Input" to?   I have nothing connected to my load breaker and I have zero export configured as you do.  But that means zero energy through the limit CTs, not zero through the inverter breaker labeled "grid".

In your case I think your inverter is drawing a small amount of power from your house at night, possibly due to losses (control electronics) or inaccuracies in the instrumentation.  Looks like about 0.5kWhr in 3 hours so that is around 130W.  Then during the day the inverter's "grid" breaker feeds around 6kWhr to your house.

Anyway that is my reading of your graphs above.

audreez, 2022-03-08:
sounds liek pbix setup is more off/grid or complete passthru grid tie, probably most of us use the ct sensors so it will be different.
im working on figuring it all out
so far i think what i/we need the most energy and power sensors for:

battery in/out
PV
essential / critical - critical load panel
non essential  - basically "grid" breaker 
total load - external CT's + essential + non essential
total grid in/out which woudl be based on CT's except in pbix scenario its based on grid breaker in sol ark itself

im almost done figuring out most of the modbus readings too will test on my system to make sure we are getting correct data and i can let you test as well

poldim, 2022-03-09:
> In my system the "grid" breaker on the inverter is the source of all the energy my system generates. From that break my whole house is supplied. So I do not understand your comment that it isn't a [...]

One of the ways of installing an Energy Storage System is to have two electrical panels.  The main electrical panel is immediately after your utility meter. It has loads attached to it that in the event of a power outage, those loads are not backed up.  One of the loads on this panel is the hybrid inverter and this is wired into the circuit breaker labeled GRID. So while it is connected to the grid, it's not immediately connected to the grid.  This type of installation is where you use the  remote-mounted CTs installed ahead of the main electrical panel alow you to measure the total grid load of your system.  The inverter then feeds a critical loads panel through the LOAD breaker.  These loads are backed up by your battery system so in the event of a power outage they will continue to work.

Here's an example of this configuration:

[image]

> In your case I think your inverter is drawing a small amount of power from your house at night, possibly due to losses (control electronics) or inaccuracies in the instrumentation. Looks like about [...]

If you look at the graph in the chart above ([here](https://github.com/pbix/HA-solark-PV/issues/7#issuecomment-1057077924)), I'm drawing 300-400w of import from the utility throughout the night, so the sol ark is not pushing power out of the grid breaker to reduce my load to 10~20w.  So that sensor shouldn't decrease.

Here is plot of solark_grid_external_total_power (the external CTs) and my industrial meter for that same time period:
[image]

pbix, 2022-03-10:
I made the below sketch with the names I propose.  I suggest you make one with the names you propose and we can then maybe compare.  

These proposed names are NOT the way to works now.  total_grid_breaker_energy is label "INV" below while "total_grid_power is labeled UTIL below.  I do not not use grid_external_total_power so I am not sure what it is.  The one labeled "main load" below is called load_power in the integration now.  Horribly confusing I know.

So there in reality 7 different power "ports" here.  Each with current, voltage, power and energy.  I do not trust that I know what the descriptions in the various MODBUS maps mean for sure.  We need to build a table that lists all 30 something sensors, what their consistent naming is AND what MODBUS register it is.   Since I do not even have a critical load panel I have no way to study that one.

[image]

poldim, 2022-03-16:
I would slightly tweak the labels as follows:
[image]

I was also looking at an energy graph that I've built for the SA:

[image]

I wonder if the fact the `inverter energy` dips below 0 is measuring it's own self-consumption.  It looks like -.7 kWh I had at 6:30 was zeroed out at 9:30.  The snaps below show that when the PV had made 1.5 kWh, about half of that went to the battery and the other half to zero out the inverter inefficiency.   

[image][image]