pbix/HA-solark-PV #8: grid buy / grid sell issue

audreez, 2022-02-27:
love this project

on my solark I use limited power to load setting most of the time and noticed that grid buy / grid sell only adds /calculates based  "inverter power"  which is then becomes the same as "load power" in my case (and probably in general) i need to calculate grid buy/ grid sell based on "grid limiters" or "grid external power" 
i can do that in HA itself but i think this would benefit someone else as well
I'm a network guy by trade but i can be a little dangerous with code, if you want me to i can attempt make some some new sensors for that or maybe you would want to redo the grid buy /sell sensors

thanks for you work!

poldim, 2022-03-02:
Not sure if you've seen the other issue that's open (#7).  It might be a naming issue where what they call GRID isn't necessarily the utility grid.

[image]

pbix, 2022-03-06:
The grid buy/sell sensors are computed by the inverter.  I think they are based on the "Limiter Sensors" shown in the figure above.  It sounds like you want them to be based on the "Load Out" breaker power.  I think "Inverter Power" is the "Grid" breaker power not the load out power as you suggest.  

Not sure what I can do for you in this case.  If you have a proposal please make a pull request for me to study.

closed 2022-03-06 by pbix