pbix/HA-solark-PV #9: fault code f58 suggestion

audreez, 2022-02-27:
can you please add fault code F58 (BMS communication fault" to the fault code list?
I unplugged my CAN comms cable from Batrium and got this code in HA
Fault Code: 0x100000000
thank you!

[image]

pbix, 2022-03-02:
The fault code you reference is for F33, AC Over Current.  When the fault bit you are looking for is active please run the following modpoll command and post the results and I will take a look.

modpoll -r 96 -c 20 <your IP address>

I must admit that I have not tested this fault string logic at all since I have not yet seen a fault on my SolArk,

audreez, 2022-03-02:
here
 
[image]

pbix, 2022-03-06:
I did a little more investigation on this issue.  First the fault screen shot you post does not seem consistent with the "official" modbus map which I attach.  F58 is listed to be something else.  This would not be the first time that SolArk documentation is confusing.  

Also it does not look like bit 57 of your modpoll output is set best I can tell.  So some undocumented behavior is going on here or I am just really confused.  Since I am not able to test these messages I am not inclined to get into a debug loop on making them work.

However, if you can modify and test the fault logic and then send me a pull request with your changes I will be happy to merge your work. 

[Modbus Sol-Ark V1.1(Public Release).pdf](https://github.com/pbix/HA-solark-PV/files/8193479/Modbus.Sol-Ark.V1.1.Public.Release.pdf)

closed 2022-03-06 by pbix