pbix/HA-solark-PV #12: Force refresh of all registers before midnight?

poldim, 2022-04-21:
Hey,

Just wondering if it's possible to force a complete pull of all of the registers shortly before midnight?  It looks like it's a bit random as to when this happens currently.

For example, in my case, I pull every t=10s.  It would be nice if the logic would say something along the lines of you need 3x pulls to ensure enough time to complete everything before midnight, so it would initiate a complete register read at `midnight - 3*t`. 

[image]

[image]

pbix, 2022-04-23:
The integration pulls all registers each period.  You define the period during configuration.  Are you thinking the integration needs to run three times to get all registers?

Your data shown above is pulled 8 seconds prior to midnight.  What do you propose exactly?

poldim, 2022-04-24:
> The integration pulls all registers each period. You define the period during configuration. Are you thinking the integration needs to run three times to get all registers?

Do you know why it's not that way on the chart above?  From 23:50, only Load and PV were updated

> Your data shown above is pulled 8 seconds prior to midnight. What do you propose exactly?

It was pulled 8 minutes before midnight, and only the PV data at that.

pbix, 2022-04-29:
Opps, I got confused about the minutes vs. seconds so sorry for that.

You can see that all registers are polled in succession in the code (hub.py) and passed as a block back to Home Assistant with a single return at the end of the function.  Not sure what more the integration can do than this.  

I suspect this is an issue with some other code somewhere in HA.  Perhaps the database or even the graphing you are using.  I would look in that direction.

closed 2022-04-29 by pbix