pbix/HA-solark-PV #55: ModuleNotFoundError

pavalenta, 2024-12-20:
Using Home Assistant via Docker with core version just updated to 2025.1.0.dev202412200230 and getting the following on start up.

Logger: homeassistant.setup
Source: setup.py:334
First occurred: 10:49:36 PM (1 occurrences)
Last logged: 10:49:36 PM

Setup failed for custom integration 'solark_modbus': Unable to import component: No module named 'pymodbus.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1034, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1014, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solark_modbus/__init__.py", line 12, in <module>
    from .hub import SolArkModbusHub
  File "/config/custom_components/solark_modbus/hub.py", line 12, in <module>
    from pymodbus.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.register_read_message'

blassond, 2025-01-04:
The following worked for me as far as pulling in DATA again, but logs indicate that it will not work on later releases.
Seems like HA updated their version of python and we will need to wait for this integration to do the same.

Temp Fix:
edit config/custom_components/solark_modbus\hub.py

Original Line 12 ------------
from pymodbus.register_read_message import ReadHoldingRegistersResponse
Updated Line 12 -------------
from pymodbus.pdu.register_read_message import ReadHoldingRegistersResponse

Original Line 64 ------------
            self._client = ModbusSerialClient(method='rtu',port=parsed.path.rstrip('/')+parsed.netloc,baudrate=9600,
Updated Line 64 -----------
            self._client = ModbusSerialClient(port=parsed.path.rstrip('/')+parsed.netloc,baudrate=9600,

pbix, 2025-01-07:
I did some investigating and found that there is once again a breaking change in pymodbus as of v3.4.7 which is now the default in HA 2025.1.  As a result this integration is now broken.  Would love to see a pull request which properly resolves this issue but if I do not I will try and get some time reserved to fix it.

In the meantime I can only advise people to refrain from upgrading to any 2025 release until a reasonable solution can be found.

poldim, 2025-01-10:
> The following worked for me as far as pulling in DATA again, but logs indicate that it will not work on later releases. Seems like HA updated their version of python and we will need to wait for [...]

thanks, this did solve the issue

poldim, 2025-01-10:
> I did some investigating and found that there is once again a breaking change in pymodbus as of v3.4.7 which is now the default in HA 2025.1. As a result this integration is now broken. Would love [...]

I created https://github.com/pbix/HA-solark-PV/pull/57 to implement @blassond's changes.  Testing this worked for me on 2015.1.2.

pbix, 2025-01-13:
Resolved at e2f1cde.
I will wait a few days for any feedback to the contrary before closing this issue.

pavalenta, 2025-01-16:
Updated HA-solark-PV in HA, forgot to upgrade HA as well. The fix appears to break in HA version 2024 (no big deal I feel), upgraded to 2025.1.0 and HA-solark-PV seems to be working as expected!

minorOffense, 2025-01-17:
Here are some logs for the newest code on the last 2024 version of HA. It seems to error. 

```
  File "/usr/local/lib/python3.13/socket.py", line 864, in create_connection
    raise exceptions[0]
  File "/usr/local/lib/python3.13/socket.py", line 849, in create_connection
    sock.connect(sa)
    ~~~~~~~~~~~~^^^^
OSError: [Errno 113] Host is unreachable[0m
[31m2025-01-17 08:27:08.543 ERROR (SyncWorker_1) [roombapy.remote_client] Can't connect to 192.168.20.6
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/roombapy/remote_client.py", line 93, in connect
    self._open_mqtt_connection()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/roombapy/remote_client.py", line 117, in _open_mqtt_connection
    self.mqtt_client.connect(self.address, self.port)
    ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/paho/mqtt/client.py", line 914, in connect
    return self.reconnect()
           ~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/paho/mqtt/client.py", line 1044, in reconnect
    sock = self._create_socket_connection()
  File "/usr/local/lib/python3.13/site-packages/paho/mqtt/client.py", line 3685, in _create_socket_connection
    return socket.create_connection(addr, timeout=self._connect_timeout, source_address=source)
           ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/socket.py", line 864, in create_connection
    raise exceptions[0]
  File "/usr/local/lib/python3.13/socket.py", line 849, in create_connection
    sock.connect(sa)
    ~~~~~~~~~~~~^^^^
OSError: [Errno 113] Host is unreachable[0m
[31m2025-01-17 08:27:08.544 ERROR (SyncWorker_1) [roombapy.remote_client] Unable to connect to 192.168.20.6[0m
```

minorOffense, 2025-01-17:
And I can confirm the latest update does work properly in the latest stable 2025 version.

pbix, 2025-01-21:
> Here are some logs for the newest code on the last 2024 version of HA. It seems to error.
>
> ```
>   File "/usr/local/lib/python3.13/socket.py", line 864, in create_connection
>     raise exceptions[0] [...]

This error does not seem to be related to the solark integration.  Error message refer to the "roombapy".  I suggest you look there for support on this problem.

pbix, 2025-01-21:
> Updated HA-solark-PV in HA, forgot to upgrade HA as well. The fix appears to break in HA version 2024 (no big deal I feel), upgraded to 2025.1.0 and HA-solark-PV seems to be working as expected!

Without an error message reported I cannot comment.  I tested in HA 2024 and it worked for me.

closed 2025-02-07 by pbix