pbix/HA-solark-PV #56: pymodbus error

Kasznaa, 2025-01-05:
Can you help me find the source of this error. the integration has been running for years it appeared after a Home assistant upgrade.

Logger: homeassistant.config_entries
Source: config_entries.py:3314
First occurred: 9:43:28 PM (1 occurrences)
Last logged: 9:43:28 PM

Error occurred loading flow for integration solark_modbus: No module named 'pymodbus.register_read_message'

Logger: homeassistant.setup
Source: setup.py:334
First occurred: 10:13:46 PM (1 occurrences)
Last logged: 10:13:46 PM

Setup failed for custom integration 'solark_modbus': Unable to import component: No module named 'pymodbus.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solark_modbus/__init__.py", line 12, in <module>
    from .hub import SolArkModbusHub
  File "/config/custom_components/solark_modbus/hub.py", line 12, in <module>
    from pymodbus.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.register_read_message'

pbix, 2025-01-06:
Can you provide information on when this starting happening and what are the previous and new version of Home Assistant versions involved?

pbix, 2025-01-07:
This is the same issue as #55 so I am closing this one.  What for a solution there.

closed 2025-01-07 by pbix

Kasznaa, 2025-01-07:
Thanks my issue does align with #55 55