pbix/HA-solark-PV #59: Modbus Error: Unable to decode frame & Modbus IO exception

poldim, 2025-01-20:
```
This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solark_modbus/hub.py:129
integration: SolArk Inverter Modbus (documentation, issues)
First occurred: January 19, 2025 at 9:49:58 PM (24506 occurrences)
Last logged: 9:16:13 AM

Unable to decode frame Modbus Error: [Input/Output] Invalid response b'\x0c\x01\x1b\x02\x19\x17p\x17p\x00\x01\x00' has byte count of 12
Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
```

I'm not sure if this is related to #55 or not, but seeing this on latest component / HA version

pbix, 2025-01-21:
Are you using a serial connection directly from your HA computer?  This is something not previously reported.

Does modpoll work?

poldim, 2025-01-28:
> Are you using a serial connection directly from your HA computer?  This is something not previously reported.
> 
> Does modpoll work?

HA container -> Modbus TCPIP Gateway -> Solark 

This is a new issue and has happened two times now with the second one being just now. Never before on any other than the current one. 
```
This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solark_modbus/hub.py:129
integration: SolArk Inverter Modbus (documentation, issues)
First occurred: 7:10:17 AM (644 occurrences)
Last logged: 7:27:57 AM

Unable to decode frame Modbus Error: [Input/Output] Invalid response b'*RE\x00\x01\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x020\x00\x00\x05P\x00\x00\x00\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00' has byte count of 42
Modbus IO exception Modbus Error: [Input/Output] Unable to decode request

```

poldim, 2025-01-30:
Another one this morning:

```

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solark_modbus/hub.py:129
integration: SolArk Inverter Modbus (documentation, issues)
First occurred: 3:33:35 AM (5138 occurrences)
Last logged: 6:07:55 AM

Unable to decode frame Modbus Error: [Input/Output] Invalid response b'\x0c\xfc\x97\xf9\x9f\x17p\x17p\x00\x01\x00' has byte count of 12
Modbus IO exception Modbus Error: [Input/Output] Unable to decode request

```

pbix, 2025-01-31:
Your report indicates that this is an intermittent issue, apparently happening about once a week.  Unfortunately, it's going to be difficult to resolve as a result.  This error is coming from the pymodbus module on which this integration depends and version of which has changed in the newest code.  I guess there is the possibility that a regression was introduced there but if so you would think there would be many reports of this issue.  

I am not aware of anything in the integration that could cause a frame decode error.  The frames are created by the inverter and passed back to HA and decoded by pymodbus.  Apparently when the frame arrives it is corrupt. 

Perhaps your poll interval could be involved?  Are you running faster than 10 second/cycle?  Have you updated your inverter firmware, possible regression might be there?  Debugging options might include using wireshark to capture all the traffic between HA and your Gateway and see if you can observe any corruption in the Modbus frames.  Then you would know which way to look from there.

Does this error cause an operational issue for you?  I would think that it would just re-poll that next cycle.  How do you become aware of these errors?

poldim, 2025-01-31:
I did have them update my fw to the latest version on 1/23. So it's a possibility that it has introduced this problem.

pbix, 2025-01-31:
Oh boy.  What version of firmware do you have in your SolArk after upgrade?  Could be the issue.

It should be possible to use the python "try/except" syntax to catch this error if it occurs and just return and thus prevent a total collapse of the integration.  You can see this syntax at line 133.   Put in the try: before the read and the expect <name>: after.  You may not need to do anything else.  Hopefully the isError() method returns true and it will just bypass that one read.

poldim, 2025-01-31:
> Oh boy. What version of firmware do you have in your SolArk after upgrade? Could be the issue.

SW Ver.: M 3.2.2.2 / S 0.8.1.7 / C 1.4.3.F

> It should be possible to use the python "try/except" syntax to catch this error if it occurs and just return and thus prevent a total collapse of the integration. You can see this syntax at line 133. Put in the try: before the read and the except : after. You may not need to do anything else. Hopefully the isError() method returns true and it will just bypass that one read.

Trying this now:
[image]

pbix, 2025-02-15:
Any more issues to report after this change?

poldim, 2025-02-15:
> Any more issues to report after this change?

No other errors since implementing this tweak, so I recommend merging it into main

pbix, 2025-02-16:
This change was made in commit [d68e001].  Hopefully it will not come back.

closed 2025-02-16 by pbix