pbix/HA-solark-PV #60: SolArk 15k - No values in HA, modpoll working

OutsmartinTech, 2025-02-04:
Home Assistant Version: 2025.1.4
Inverter: Sol-Ark 15K-2P-N

I have setup my device as per the installation guide and am able to get results from using modpoll. However, I cannot figure out how to get the integration working and am unsure what the issue is.

My RS485 to TCP bridge is at 10.0.0.34, port 502.
When testing with modpoll, I had to use slave id 0.

When setting up the integration, I tried using 10.0.0.34 as the host, which never retrieved any values (all sensor entities show 'Unknown').
I removed the integration entry and re-added with a host of 10.0.0.34/;0 which never retrieved any values.
I then removed the integration entry again and re-added with a host of 10.0.0.34:502/;0 which also never retrieves any values.

I checked the logs section in the HA front end and was unable to see any log lines pertaining to this integration.

I'm probably just doing something wrong, any suggestions?

OutsmartinTech, 2025-02-04:
Figured it out, line 92 in hub.py doesn't consider 0 to be a valid slave id. I modified the hub.py on my system as follows.

Changed:
`        if (parsed.params.isdigit()) and (int(parsed.params) > 1) and (int(parsed.params) < 256):`

to
`        if (parsed.params.isdigit()) and (int(parsed.params) < 256):`

Rebooted the HA server, and all is working now. Thanks for a great integration, this is awesome!

pbix, 2025-02-16:
This change was made in commit [d68e001].

closed 2025-02-16 by pbix