pbix/HA-solark-PV #61: HA 2025.02 released today, another pymodbus upgrade

OutsmartinTech, 2025-02-06:
I installed the new HA update today, which breaks this custom component again due to upgrading to v3.8.3 of pymodbus. I inserted the following into the middle of the if/else section that starts at line 27 in hub.py. This gets the component to load again. However, there must be another change as well, since all my values are currently showing "Unavailable" now.

```
elif hasattr(pymodbus,'__version__') and ((pymodbus.__version__[0] == '3') and (pymodbus.__version__[2] >= '8')):
   from pymodbus.pdu.register_message import ReadHoldingRegistersResponse
```

I will continue to fiddle and if I figure out anything else, will post here. I'm no python expert, so I'm not making any promises.

OutsmartinTech, 2025-02-06:
Figured the other issue out as well (I think). pymodbus has changed the read_holding_registers function to force several of the params to be by keyword only. In hub.py, changing the return for the _read_holding_registers function to:

`            if hasattr(pymodbus,'__version__') and ((pymodbus.__version__[0] == '3') and (pymodbus.__version__[2] >= '8')):
                return self._client.read_holding_registers(address=address, count=count, slave=unit)
            else:
                return self._client.read_holding_registers(address, count, **kwargs)
`

fixed the issue, and I'm getting values now. There's another message in the log now that states the following:

`BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 ! Please use "client.convert_from_registers()" or "client.convert_to_registers" See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"`

It's working for now, so I'm ignoring that warning message.

pbix, 2025-02-12:
I will work on this this weekend 2/21, thanks for the report.

Vartkat, 2025-02-13:
@OutsmartinTech I did your modifications in hub.py, it loads and I got my values.

Thanks

ruralsysadmin, 2025-02-13:
@OutsmartinTech same here as @Vartkat , appears to be working as expected with the referenced error message repeating. Thanks for figuring this out!

pbix, 2025-02-16:
Both issues identified above are fixed in commit [d68e001].  Thanks for helping by researching the root cause.

closed 2025-02-16 by pbix