pbix/HA-solark-PV #63: HA 2025.2.4 appears to break again?

dlasher, 2025-02-16:
Already running fixes in https://github.com/pbix/HA-solark-PV/issues/61 - updated from 2.2 to 2.4, breaks again. Verified i'm running current code in github - checking here : https://github.com/pbix/HA-solark-PV/commit/d68e001ac15d1af1b331de64e78e26b853b308ce

`Core
2025.2.4
Supervisor
2025.02.1
Operating System
14.2
Frontend
20250214.0`

have re-pulled from github, rebooted, etc. No love. :(

Errors below:

Log details (ERROR)
Logger: homeassistant.setup
Source: setup.py:340
First occurred: 11:02:31 PM (1 occurrences)
Last logged: 11:02:31 PM

Setup failed for custom integration 'solark_modbus': Unable to import component: No module named 'pymodbus.pdu.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 340, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solark_modbus/__init__.py", line 12, in <module>
    from .hub import SolArkModbusHub
  File "/config/custom_components/solark_modbus/hub.py", line 30, in <module>
    from pymodbus.pdu.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.pdu.register_read_message'

zeroping, 2025-02-16:
I've been watching #61 closely, and also tried [d68e001], and got the same error.

I appear to be on pymodbus 3.8.3, and needed this change in [hub.py](https://github.com/pbix/HA-solark-PV/blob/d68e001ac15d1af1b331de64e78e26b853b308ce/custom_components/solark/hub.py#L32):

```
if hasattr(pymodbus,'__version__') and ( ((pymodbus.__version__[0] == '3') and (pymodbus.__version__[2] <= '1')) or (pymodbus.__version__[0] <= '2')):
   from pymodbus.register_read_message import ReadHoldingRegistersResponse
   prior347 = True
else:
   from pymodbus.pdu.register_read_message import ReadHoldingRegistersResponse
```
to
```
if hasattr(pymodbus,'__version__') and ( ((pymodbus.__version__[0] == '3') and (pymodbus.__version__[2] <= '1')) or (pymodbus.__version__[0] <= '2')):
   from pymodbus.register_read_message import ReadHoldingRegistersResponse
   prior347 = True
elif hasattr(pymodbus,'__version__') and ((pymodbus.__version__[0] == '3') and (pymodbus.__version__[2] < '8')):
   from pymodbus.pdu.register_read_message import ReadHoldingRegistersResponse
else:
   from pymodbus.pdu.register_message import ReadHoldingRegistersResponse
```

I haven't followed pymodbus closely enough to understand when/why `pymodbus.pdu.register_read_message` changed to `pymodbus.pdu.register_message`, but supposedly the API is stable across all of 3.8.x. Maybe I'm just on a different pymodbus version than #61 was tested against?

dlasher, 2025-02-16:
Good catch - inserting the ELIF in-between.... and it failed.

copy/pasted the entire (7) lines. Restarted, and.... 

and it worked for me, thank you!

> SolArk Grid Relay changed to Closed
> 11:30:01 AM - 4 minutes ago
> SolArk Inverter error message changed to ['No Faults']
> 11:30:01 AM - 4 minutes ago

OutsmartinTech, 2025-02-16:
Yeah, I think a couple of the fixes were missed in the commit. I upgraded as well (must have the newest version of all the things!) and had to make the change identified as above, as well as the one specific to my Solark model to allow slave id 0.

pbix, 2025-02-17:
Both corrections made at a53dfa9, please check again.
Sorry guys, thanks for your patience and help.  Clearly, this is not my day job.

closed 2025-02-17 by pbix