pbix/HA-solark-PV #64: Sol-ark entities unavailable after power outage

Tigeruno, 2025-03-04:
Lost grid, came back on in about 15 minutes.  The circuit that the HA Green is on was not covered by transfer box, so it lost power.
I deleted dashboard, integration and tried to re-install.  Download does not work as the integration exists in HACS store.
All the py files appear to be in place?

The Solark is connected via rs-485 using DSD TECH SH-U11F USB-RS485 on /dev/ttyUSB1.  Tested cat5 again with modpoll and all solark data is coming through.  I'm stumped!

pbix, 2025-03-05:
Please post the relevant log file section.Perhaps there are some hints there.Sent from my iPhone

Tigeruno, 2025-03-05:
> Grid down.  My last try, deleted hacs dir where solar files were.  No go.
> Will get log files soon.
> lost power. <br> I deleted dashboard, integration and tried to re-install.
> Download does not work as the integration exists in HACS store. <br> All
> the py files appear to be in place? <br> The Solark is connected via rs-485
> is on was not covered by transfer box, so it lost power. <br> I deleted
> the integration exists in HACS store. <br> All the py files appear to be in
> place? <br> The Solark is connected via rs-485 using DSD TECH SH-U11F
> data is coming through. I&#39;m stumped! <br> <br> —Reply to this email

Note:  esphome is setup and working (JK_BMS) but not connected during this
time.

Tigeruno, 2025-03-05:
HACS configs

tommywatson, 2025-03-05:
Is your BMS running? I had an issue yesterday where the power went out and someone in the house used the hot water (insta-hot) and it tripped one of the breakers. My installer wired it to the two center posts, requiring both breakers to be on, not just one, until the second breaker was turned on, the BMS was out and I was getting erratic readings from the SolArk in HA.

Good luck!

Tigeruno, 2025-03-05:
My DIYBMS and JK_BMS are not connected to sol-ark8k, so not the issue.
Modpoll shows good data coming from sol-ark.

Tigeruno, 2025-03-06:
Im up and running with Sol-Ark.  Have a lot to learn!  Thanks for any help,
much appreciated!

closed 2025-03-15 by pbix