pbix/HA-solark-PV #66: Fails to automatically connect to tcp rs485 device

hansaya, 2025-04-23:
First of all, thank you for all the hard work. I love this project! I do have a small ask, is there any way to gracefully handle a disconnect from the rs485 tcp device? Currently when this integration losses connection from device, it errors out and never attempts to connect back. Unless you reload the integration.

```
This error originated from a custom integration.

Logger: custom_components.solark_modbus.hub
Source: helpers/update_coordinator.py:380
integration: SolArk Inverter Modbus (documentation, issues)
First occurred: 12:53:56 AM (7 occurrences)
Last logged: 12:56:56 AM

Unexpected error fetching SolArk-test data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solark_modbus/hub.py", line 296, in _async_update_data
    realtime_data = await self.hass.async_add_executor_job(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self.read_modbus_realtime_data
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/solark_modbus/hub.py", line 326, in read_modbus_realtime_data
    realtime_data = self._read_holding_registers(unit=self.slaveno, address=60, count=21)
  File "/config/custom_components/solark_modbus/hub.py", line 280, in _read_holding_registers
    return self._client.read_holding_registers(address, count=count, slave=unit)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 114, in read_holding_registers
    return self.execute(no_response_expected, pdu_reg.ReadHoldingRegistersRequest(address=address, count=count, dev_id=slave))
           ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 203, in execute
    return self.transaction.sync_execute(no_response_expected, request)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 119, in sync_execute
    raise ModbusIOException(txt)
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request
```

pbix, 2025-04-26:
I agree this is a problem and I suffer the problem.  A reload is required to re-establish a broken connection.  I do not know the best way to resolve it and invite others to contribute a solution.

The strange thing to me is that I do not remember this being an issue before and wonder if some update in Home Assistant caused this.

ruralsysadmin, 2025-04-26:
Agreed on don't recall seeing this before, don't remember when it started though either. Don't have coding experience but happy to try apply anything to provide feedback

dcastenholz, 2025-04-26:
I am going to take a stab at this.  40 years of software experience, but 0 Python, so YMMV.

hansaya, 2025-05-20:
@dcastenholz did you get a chance to take a look at this?

dcastenholz, 2025-05-20:
I’ve been unexpectedly busy and have not had a chance to finish my reconfiguration workflow code yet. I am still working on it though. I will take a look at the communication failures after that, but I’m not expecting to be able to get to it in the next two weeks at least.

minorOffense, 2025-10-02:
I just started getting this error recently. Normally a quick reload of the service resolves it but now it's just stuck (haven't collected data for a week now). Next test step I'll go power the actual raspizero off/on again. See if that does anything.

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 392, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solark_modbus/hub.py", line 297, in _async_update_data
    self.inverter_data = await self.hass.async_add_executor_job(
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self.read_modbus_inverter_data
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/solark_modbus/hub.py", line 312, in read_modbus_inverter_data
    inverter_data = self._read_holding_registers(unit=self.slaveno, address=3, count=5)
  File "/config/custom_components/solark_modbus/hub.py", line 285, in _read_holding_registers
    return self._client.read_holding_registers(address, count=count, device_id=unit)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/mixin.py", line 113, in read_holding_registers
    return self.execute(no_response_expected, pdu_reg.ReadHoldingRegistersRequest(address=address, count=count, dev_id=device_id))
           ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/client/base.py", line 202, in execute
    return self.transaction.sync_execute(no_response_expected, request)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 153, in sync_execute
    raise ModbusIOException(txt)
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request
2025-10-01 19:53:51.074 DEBUG (MainThread) [homeassistant.components.govee_light_local.coordinator] Disconnected
2025-10-01 19:54:24.850 ERROR (SyncWorker_16) [pymodbus.logging] No response received after 3 retries, continue with next request
2025-10-01 19:54:24.850 ERROR (MainThread) [custom_components.solark_modbus.hub] Unexpected error fetching SolArk data
```

benbatya, 2026-02-11:
After restarting the system, the data is collecting again.

But I expect it to fail with the same transaction_id error soon.
Thanks