pbix/HA-solark-PV #68: No module named 'pymodbus.register_read_message'

sbb33, 2025-05-14:
Hi, 

I've been using this integration since last August. Really appreciate all the work that has been put into this.

I just updated to latest HAOS, Core, and HACS. Everthing updated fine except for the SolArk Inverter Modbus Integration which says "Not loaded".

Sorry, I'm kind of in panic mode. If this has been covered or I need to provide more information please let me know.

Thank You

Here is the error log.

Logger: homeassistant.setup
Source: setup.py:344
First occurred: 6:56:29 PM (1 occurrence)
Last logged: 6:56:29 PM

Setup failed for custom integration 'solark_modbus': Unable to import component: No module named 'pymodbus.register_read_message'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 344, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1013, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 993, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1053, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solark_modbus/__init__.py", line 12, in <module>
    from .hub import SolArkModbusHub
  File "/config/custom_components/solark_modbus/hub.py", line 12, in <module>
    from pymodbus.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.register_read_message'

sbb33, 2025-05-14:
Sorry just noticed it covered in #55.

sbb33, 2025-05-14:
Hi,

After some reading, I'm not sure how to access the code that I need to edit.
Also, not sure which fix I should use...there seems to be some updates since #55.
Thanks

sbb33, 2025-05-14:
I used my back up to successfully restore full functionality.

I will look into updating at a later date.

Thanks again for creating and maintaining this application!

pbix, 2025-05-16:
> Hi,
> 
> After some reading, I'm not sure how to access the code that I need to edit. Also, not sure which fix I should use...there seems to be some updates since [#55](https://github.com/pbix/HA-solark-PV/issues/55). Thanks

Issue #55 is fixed and closed so I suspect this issue not related to that one.

What version of HA are you running and are you using the latest version for the Solark integration?  This is the first report of this problem that I am aware of.

sbb33, 2025-05-16:
Hi Paul,

Thank you for following up.

I am using home assistant green and I updated to the current Home assistant
operating system 15.2. Then updated to Core 2025.5.1. After that I updated
to Hacs 2.0.5.
There was no update shown for the Solark Integration and all data fields
said unknown or unavailable, can't remember which at the moment. I checked
the logs and it showed the error I previously provided.

I used my backup to roll back to core 2024.12.1 and Hacs 2.0.1 and it
started working again.

Please let me know if you need more information.

Thanks

sbb33, 2025-05-16:
Also, I am current using solark integration accc938.

pbix, 2025-05-24:
The current version of this integration as of this writing is "a53dfa9".  
It does not contain the line referenced in your error message above:
 
`File "/config/custom_components/solark_modbus/hub.py", line 12, in
from pymodbus.register_read_message import ReadHoldingRegistersResponse
ModuleNotFoundError: No module named 'pymodbus.register_read_message'`

Today I upgraded my system to the following and observed no issues.

Installation method Home Assistant OS
Core 2025.5.3
Supervisor 2025.05.1
Operating System 15.2
Frontend 20250516.0

I suggest you upgrade to the latest version of this integration and match all the above versions.

As I cannot reproduce the reported issue I am closing this.

closed 2025-05-24 by pbix

sbb33, 2025-05-24:
Thank you for checking into this and following up.
Will the latest version of this integration show up in the update list after updating home assistant?
It did not show up for me after updating home assistant previously.