pbix/HA-solark-PV #71: Continuation of Closed #68

sbb33, 2025-05-28:
Hi, this is a continuation of #68 which has been closed as stated below by pbix:

"The current version of this integration as of this writing is "a53dfa9".
It does not contain the line referenced in your error message above:

File "/config/custom_components/solark_modbus/hub.py", line 12, in from pymodbus.register_read_message import ReadHoldingRegistersResponse ModuleNotFoundError: No module named 'pymodbus.register_read_message'

Today I upgraded my system to the following and observed no issues.

Installation method Home Assistant OS
Core 2025.5.3
Supervisor 2025.05.1
Operating System 15.2
Frontend 20250516.0

I suggest you upgrade to the latest version of this integration and match all the above versions.

As I cannot reproduce the reported issue I am closing this."

I realize that I am on an old version of this integration. My problem is I can not update to the latest version of this integration.
There are no updates available before or after updating all of the home assistant areas.
Sorry, I did not make this clear in #68.

Thanks

pbix, 2025-07-18:
Seems this issue is resolved so I am closing it.

closed 2025-07-18 by pbix

sbb33, 2025-07-18:
I still have this issue, but I agree with your decision to close it.

Thanks for developing and sharing this integration.